UsbCANConverter-UCCbasic/UCCBEmbedded #1: UART pins TX

1am, 2017-11-08:
Hello,

I'm experiencing an issue with 2 USBCanConverterBasic devices I've tested.
So far I've been able to isolate the issue described below:

When the device is powered from USB I can normally communicate with the board via USB virtual serial port. 

While the device is powered via USB and while using UART through an additional USB-UART adapter (3V3 logic levels) I can send commands to USBCanConverterBasic but output is sent to USB virtual serial port and nothing is transmitted back to my USB-UART adapter's port.

When I power the device only via 5V pin (USB not connected to anything) I can send the command through USB-UART but i get a response only once. All subsequent writes don't return anything which is visible in the attached screenshot (first row is my data being sent, second is the response. In both cases I've sent "v").

[image]

Looking at the sources I think it's caused by this part https://github.com/UsbCANConverter-UCCbasic/UCCBEmbedded/blob/master/Src/slcan/slcan.c#L125 but since I've never written a line on STM I'm not sure. 

Wouldn't it be easier to just broadcast responses to both USB and UART pins?

Can you advise on how to fix the situation or what could be wrong potentially?

Lukasz-Juranek, 2017-11-08:
Hi, UART sending was not tested much, so there can be some issues.

1. Regarding not sending via UART when USB is connected. 
Yes this is done on purpose,  removing  `if (hUsbDeviceFS.dev_state != USBD_STATE_CONFIGURED)` will enable simultaneous sending on both interfaces. But i would not recommend that, UART is typically max 115200 that is not enough to pass all data from High speed CAN, You can try with higher speed but this will probably introduce more errors on physical layer of UART. Also 'HAL_UART_Transmit' is blocking function so You will miss some CAN frames. Also sensing via UART is not error proof so You will probably need to implement some additional checksum mechanism or use parity bits from UART, that are so so.

2. Regarding 
> only one response issue

this is bug probably, I need more time to investigate this. 

If You are trying to use UART interface to communicate with other microcontroller then You should consider rewriting SLCAN interface for something else with less data overhead and some checksums.

Regards,

Lukasz-Juranek, 2017-11-08:
I've added pull request with possible solution to Your problem, it is untested, if You have time please check it, there are release [hex and dfu files](https://github.com/UsbCANConverter-UCCbasic/UCCBEmbedded/releases/tag/v0.1)  for this pull request available #2 . Uart transmission will work only if USB is unconfigured state.

1am, 2017-11-08:
Thank you. I'll try to test tomorrow.

1am, 2017-11-09:
Hi. After doing DFU it seems that the issue is still there. I type the "v" command and get response only once.

Lukasz-Juranek, 2017-11-13:
Ok, thanks for testing, I've done some testing too, and in My setup all works well, even on software without "fix". I've done test on UCCB HW 0101, FW 0203. My setup is as follows CP2102 usb to UART converter connected to PC and from it, all 4 pins (GND RX TX VCCIO) connected to UCCB. I've got 5 Volts setup on VCCIO. UCCB USB connector is disconnected.

closed 2019-10-21 by Lukasz-Juranek