UsbCANConverter-UCCbasic/UCCBEmbedded #3: OpenWRT s issue

riodda, 2018-02-11:
Seems that on openwrt the s parameter on slcand prevents the slcand demon to load the can interface, no error messages are received.
if is issue:
`slcand -o -c -f /dev/ttyACM0 slcan0`
and 
`ifconfig slcan0 up`
all works ok i read the traffic on the canbus, on the other hand if i issue
`slcand -o -c -f  -s8 /dev/ttyACM0 slcan0`
and 
`ifconfig slcan0 up`
I don't see any traffic, i don't know if the issue is related to slcand on openwrt only.
i've used your firmware on a slight modified interface based on minipci express format.

Lukasz-Juranek, 2018-02-12:
If You have some time, you can debug this issue using program like
interceptty <https://github.com/Lukasz-Juranek/interceptty> , or other
software for serial port monitoring. See what is really send on  USB
virtual serial port.

But probably This is issue with dongle, there is a bug for handling
multiple commands send in one USB packet, only first command will be
processed. This is the case when commands from lscand are send without any
delay.

If this is the issue in Your case, This can be fixed in two ways.

In slcand sources in slcand.c file there is:

if (speed) {
sprintf(buf, "C\rS%s\r", speed);
write(fd, buf, strlen(buf));
} This code assumes that slcand compatible device can handle more then one
message in one command. This is not true for UCCB :)

Recompile slcand with multiple write  both for close and speed and  a small
delay before sending speed parameter.

This should fix the problem.

Or fix UCCB codes to handle up to 2 commands in one USB packet.

Regards.
LJ

riodda, 2018-02-12:
Thanks, if i send the commands manually with minicom all works flawless, i think your hipotesys for multiple commans if probably the right one, will try to tweak the slcand code.
Cheers.

macpijan, 2018-03-16:
@riodda @Lukasz-Juranek 
You may want to try building from this branch: https://github.com/vitroTV/can-utils/commits/fix_read_cr_after_command
We had the same issue a while ago and solved it independently.

iwasz, 2019-10-03:
Hi all, I stumbled across the same problem, but I am running newest Manjaro on x86. I'll write down what I've learned and maybe it will be of help to someone. So my UCCB had firmware version 0303 if I remember correctly and the only way of making slcand to work was to :

```sh
sudo java -jar uCCBViewer-2.2.jar
```
Then you would set desired baudrate (in my case 500kbps), close the app immediately and then you do exactly what @riodda said :
```
sudo slcand -o -c -f /dev/ttyACM0 slcan0
sudo ifconfig slcan0 up
```
Issuing ```candump slcan0``` lets you verify if the interface works.

But If one wants to get rid of this "manual baudrate setting" step, the firmware has to be upgraded. @Lukasz-Juranek got it covered in the docs, but the only thing he forgot to mention is that you have to convert intel hex files he provides to binary format. It got me puzzled as I thought that dfu-util recognizes the format of a file and I tried to upload 0401.hex with no luck. You simply :

```objcopy -Iihex -Obinary 0400.hex 0400.bin```.

closed 2019-10-21 by Lukasz-Juranek