UsbCANConverter-UCCbasic/UCCBEmbedded #6: Ramka w trybie aktywnym wysyłana jest w pętli

rdbv, 2018-07-11:
Wersja FW0204/HW0101
Przy wysyłaniu ramki w trybie aktywnym nadajnik wpada w pętle i wysyła ją bez przerwy - sprawdzone na oscyloskopie.
W trybie loopback problem nie występuje.
Sprawdzone za pomocą UCCBViewer 1.4 oraz zwykłych komend po UART.
Obstawiam że to problem w sofcie, jeszcze nawet do kodu nie zerkałem :D, gdyż UCCB dostałem dopiero dziś. 
W zasadzie jeśli jedynym wynikającym z tego problemem jest to że muszę używać trybu loopback, to nie jest to duży problem. 
Konfigurując interfejs w Linuxie również nic nie jest wysyłane gdy odpalę `cangen slcan0`, ale linuxowy interfejs na razie zostawiam na boku gdyż używać będę interfejsu do zabaw z moim samochodem. 
Dobrym pomysłem było by też żeby dać 2 piny w miejsce gdzie trzeba lutować żeby zmienić opór rezystora, o wiele wygodniejsze.
Pozdrawiam, i tak ogólnie, to całkiem spoko sprzęt.

Lukasz-Juranek, 2018-07-11:
1. Czy ma Pan coś podłączone do urządzenia od strony magistrali CAN ? Jeśli
nie to własnie to jest przyczyną ciągłego wysyłania ramki, ramka będzie się
tak długo wysyłać aż żadne inne urządzenie jej nie potwierdzi. Jest to
jeden z dwóch trybów w jakich może działać nadajnik ramek CAN, nie da się
tego zmienić z poziomu komputera, można tylko wyedytować źódła żeby to
zmienic. Natomiast problem zniknie jak połączy pan drugie urządzenia do
magistrali.

2. canutils te domyślne nie działa za dobrze z tym urządzeniem, ponieważ
wysyła kilka komunikatów slcan w jednej paczce, proszę spróbować
przekompilować sobie can-utils z tych źródeł jest tam kilka bugów
poprawionaych

git clone https://github.com/vitroTV/can-utils.git

3. można też zrobić poprawke w kodzie UCCB zgodnie z tym pull-requestem

https://github.com/UsbCANConverter-UCCbasic/UCCBEmbedded/pull/5

ale nie gwarantuje ze to bedzie działać bez zarzutu, bo nie sprawdzałem.

Pozdrawiam
JŁ

rdbv, 2018-07-11:
Połączyłem UCCB z shieldem CAN dla Arduino i ma Pan całkowitą rację. Mój błąd.
Dziękuję bardzo za szybką odpowiedź i pozdrawiam.

closed 2018-07-11 by rdbv