UsbCANConverter-UCCbasic/UCCBEmbedded #7: After upgrade to latest firmware UCCB is detected as STM32 Bootloader

rdbv, 2018-07-12:
I needed 84000 CAN speed that was available in latest firmware for connecting with Mercedes CAN. 
So i installed 0302 version.

Installation was exacly the same as in https://ucandevices.github.io/uccb.html instructions and ended succesfully, but after reconnecting device, is still detected as STM32 Bootloader.

- Log from dmesg

[63352.114194] usb 1-1.1.4: USB disconnect, device number 56
[63352.848162] usb 1-1.1.4: new full-speed USB device number 57 using ehci-pci
[63352.957198] usb 1-1.1.4: New USB device found, idVendor=0483, idProduct=df11
[63352.957206] usb 1-1.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[63352.957211] usb 1-1.1.4: Product: STM32  BOOTLOADER
[63352.957215] usb 1-1.1.4: Manufacturer: STMicroelectronics
[63352.957219] usb 1-1.1.4: SerialNumber: FFFFFFFEFFFF

I think that this is something simple, but this is my first contact with STM, and i don't want to accidentaly brick my UCCB.
Thanks in advance! :)

rdbv, 2018-07-12:
I did everything exactly as in the instructions, after upload, I pressed the 'Leave DFU mode' button.

Lukasz-Juranek, 2018-07-12:
> 2018-07-12 11:27 GMT+02:00 Łukasz Juranek <juranek.lukasz@gmail.com>:

Sorry for multiple mails I've got problems controlling my mailbox :) .

Case You did so please try again, i'll try to post video on how i'm doing
this probably next week.

Regards
LJ

2018-07-12 11:27 GMT+02:00 Łukasz Juranek <juranek.lukasz@gmail.com>:

Lukasz-Juranek, 2018-07-12:
Ok then this is strange behavior that should not occur. Please try again
doing it.

Lukasz-Juranek, 2018-07-12:
There are also tools for dfu under linux please see

http://dfu-util.sourceforge.net/dfu-util.1.html

http://dfu-util.sourceforge.net/dfuse.html

2018-07-12 11:29 GMT+02:00 Łukasz Juranek <juranek.lukasz@gmail.com>:

rdbv, 2018-07-12:
Yes, I knew that it was on Linux (and i needed to put out my old rusty unused Windows PC from closet), but I was not sure what commands i should use (memory, etc) to upload it with Linux dfuse, can You show how to upload FW from Linux?

Lukasz-Juranek, 2018-07-12:
I'll try do give You the answer tomorrow i need to make some testing.

Lukasz-Juranek, 2018-07-12:
Ok so i've done some testing to upload 0204 firmware i'v done

*sudo dfu-util -l*

*wget
https://github.com/UsbCANConverter-UCCbasic/UCCBEmbedded/releases/download/2.4/0204.dfu
<https://github.com/UsbCANConverter-UCCbasic/UCCBEmbedded/releases/download/2.4/0204.dfu>sudo
dfu-util -a 0 -D 0204.dfu *

*sudo dfu-util -a 0 -s 0x08000000:leave -U ~/dummy*

this command is workaround to leave dfu mode, it reads memory contests to
dummy file, and leaves dfu mode.

If device will not exit for first time try last command once again.

Regards
LJ

2018-07-12 11:54 GMT+02:00 Łukasz Juranek <juranek.lukasz@gmail.com>:

rdbv, 2018-07-13:
OK, it works. Thanks. (Uploaded 0302 version).
Also i will try to build a compiler for this STM and experiment a bit with code, great that the entire UCCB is open-source.
Keep up good work and thanks for fast help!

closed 2018-07-13 by rdbv