UsbCANConverter-UCCbasic/UCCBEmbedded #8: Better explanation of filter command

a1lu, 2019-09-18:
Hey,
Could you please provide a better explanation how the filter command works?
I don't understand anything of the provided explanation.

What are filter and bank numbers?
Which bits are used for `filter flags`, e.g. for Mode: is `0` Mask or Filter?
Which scale mode is set if `1`?
what to set in iiiiiiii and mmmmmmmm

Thanks in advance and best regards

Lukasz-Juranek, 2019-09-19:
Hi,
1. To test filters please check UCCBViewer and sniff what is sended via
virtual COM port.
2. For explenation please see microcontorller manual
<https://www.st.com/content/ccc/resource/technical/document/reference_manual/c2/f8/8a/f2/18/e6/43/96/DM00031936.pdf/files/DM00031936.pdf/jcr:content/translations/en.DM00031936.pdf>,
search for bxCAN (page 814)
<https://www.st.com/content/ccc/resource/technical/document/reference_manual/c2/f8/8a/f2/18/e6/43/96/DM00031936.pdf/files/DM00031936.pdf/jcr:content/translations/en.DM00031936.pdf>

If this is not sufficient please let me know

a1lu, 2019-09-19:
Hi Lukasz,
thanks for your quick reply. I don't understand exactly what's happening with the filters in the data sheet, but for me it's currently enough to set single id filters.

I used `socat` to sniff the port (https://unix.stackexchange.com/a/225904) and got the following results:
```
M00007000000010000000010
M01017000000020000000020
M02027000000030000000030
M03037000000040000000040
M04047000000050000000050
```
These are filters for identifier 001 to 005 in the banks 00 to 04.

closed 2019-10-21 by Lukasz-Juranek