UsbCANConverter-UCCbasic/UCCBEmbedded #9: selected processor does not support `dsb 0xF' in ARM mode

jshxcn, 2019-09-24:
hi，Lukasz-Juranek 
    When I was building the project, there was an error. Can you help me solve this problem? I have not used this IDE before.
    The error is as follows:
arm-atollic-eabi-gcc -c -mthumb -mcpu=cortex-m0 -std=gnu11 -D__weak=__attribute__((weak)) -D__packed=__attribute__((__packed__)) -DUSE_HAL_DRIVER -DSTM32F042x6 -I../Inc -I../Drivers/STM32F0xx_HAL_Driver/Inc -I../Drivers/STM32F0xx_HAL_Driver/Inc/Legacy -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F0xx/Include -I../Drivers/CMSIS/Include -ffunction-sections -fdata-sections -g -Wall -o Src\usbd_desc.o ..\Src\usbd_desc.c 
arm-atollic-eabi-gcc -c -std=gnu11 -D__weak=__attribute__((weak)) -D__packed=__attribute__((__packed__)) -DUSE_HAL_DRIVER -DSTM32F042x6 -I../Inc -I../Drivers/STM32F0xx_HAL_Driver/Inc -I../Drivers/STM32F0xx_HAL_Driver/Inc/Legacy -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F0xx/Include -I../Drivers/CMSIS/Include -ffunction-sections -fdata-sections -g -Wall -o Drivers\STM32F0xx_HAL_Driver\Src\stm32f0xx_hal_cortex.o ..\Drivers\STM32F0xx_HAL_Driver\Src\stm32f0xx_hal_cortex.c 
C:\Users\jshxcn\AppData\Local\Temp\ccFEasoe.s: Assembler messages:
C:\Users\jshxcn\AppData\Local\Temp\ccFEasoe.s:431: Error: selected processor does not support `dsb 0xF' in ARM mode
C:\Users\jshxcn\AppData\Local\Temp\ccFEasoe.s:446: Error: selected processor does not support `dsb 0xF' in ARM mode

Lukasz-Juranek, 2019-09-24:
Please specify what IDE are you using?

I did not faced this problem,  I'm using Atollic TrueSTUDIO® for STM32
Version: 9.1.0  <https://atollic.com/resources/download/> please try with
this IDE.

jshxcn, 2019-09-25:
I updated the IDE and there were no problems.
Thanks

closed 2019-10-21 by Lukasz-Juranek