UsbCANConverter-UCCbasic/UCCBEmbedded #10: Problems sending a frame

iwasz, 2019-10-06:
I have had serious problems with sending CAN frames reliably with the device, the other part (listening to it) wouldn't receive anything, even though both UCCBViewer and slcan driver reported that sending was successful (I tried both the app and can-utils tools). After inspecting the device with logic analyzer I found, that the other part isn't indeed receiving frames because they are sent with incorrect ID and thus its filter was filtering them out. Here you have a screenshot of a frame which had to have ID == 0x12345678, DLC == 8, and data == 0x11, 0x22, 0x33 etc.

http://iwasz.pl/files/uccb-2.png - notice how timing is off

The worst part is, that the ID changed over time. Sometimes I saw 0x12345670, sometimes 0x123454F0 etc. I spent quite some time debugging this as I badly needed (and still do) a working device. I suspected stack overflow, because watchdog (that I didn't noticed was on) deceived me resetting the CAN_HandleTypeDef structure to zeroes (watchdog works under debugger - I didn't know that either). To make long story short, the problem lies (IMHO) in the clock configuration, and further in the hardware itself (my HW was bought in the botland store and reports 0102 version). 

You derive your main clock from the internal clock HSI48 which drives USB. I don't know what is it's accuracy, but switching SYSCLK to HSI + PLL improved the situation a little, but not quite. I noticed that with that configuration, last bytes of payload were corrupted. And then I soldered 16MHz crystal (with 10pF loads), and this definitely fixed the problem : 

http://iwasz.pl/files/uccb-3.png

Remarks:
1. Receiving works just fine.
2. I tried firmware versions 0303, 0400 and 0401 on HW version 0102
3. I see, you are running out of FLASH, and thus you commented parts of HAL_RCC_OscConfig function and timeout stuff in HAL_CAN_Transmit (called from transmitStd). This is a serious problem if you want to compile without -Os like I did to be able to inspect stuff under the debugger. I at the other hand commented out the bootloader, the watchdog, and bitrate stuff (I need only 500kbps). I think you can get away from this compiling with -Os or -O3 when releasing, although it didn't worked for me (the device hanged).
4. Other than adding a crystal, I think you could try to provide some means of calibration to your device. I think that it should be doable.
5. I really like your device, and I really appreciate it is open sourced, which enabled me to investigate. 
6. Internal oscillators are susceptible to temperature and other external conditions, so I think this is why you haven't noticed any problems. 
7. I commited my changes to https://github.com/iwasz/UCCBEmbedded/tree/extid-problems , but since you are not using an external oscillator, they are of no use for anyone but me for now.

Please let me know if you see some other means of fixing this.

Lukasz-Juranek, 2019-10-07:
I really appreciate amount of work you've done. Thanks for feedback is very
valuable.

1. In What conditions do You use this device ?
2. Did this issue occurred in more then one pcs of Hardware ?
3. Did you tried also GS_USB
<https://github.com/UsbCANConverter-UCCbasic/UCCB_GS_Embedded/releases>software
for this ?

Regards
LJ

iwasz, 2019-10-07:
Hi

> 1. In What conditions do You use this device ?

Simply in my "lab", temperature around 20°C, average humidity. Device
connected through an USB extension cord, then twisted pair goes to the
device our team is working on.  Logic analyzer (added later on) is
sniffing signals on the pin 1 (TX I believe) of the transceiver on
UCCB.

> 2. Did this issue occurred in more then one pcs of Hardware ?

Unfortunately I have only one of these.

> 3. Did you tried also GS_USB
> <https://github.com/UsbCANConverter-UCCbasic/UCCB_GS_Embedded/releases>software
> for this ?

I don't think I follow. I used dfu-util at first, and flashed 0301,
0400 and 0400 with the same results (thanks to you I learned about
this great tool, and will be using it in the future for sure). Then I
switched to normal st-link (soldered the programming header). Oh, do
you want me to try FW 0002 version?

Regs.

Lukasz-Juranek, 2019-10-07:
Three is seperate software for Linux only that is not using virtual com
port but Linux gs_usb driver it is more robust soltuion, so if You are
using Linux use gsusb software instead of standart one, You   can upload it
using USB bootlader, as described on link.

I'll send You another device for testing free of charge send me Your
shiping addres on    ucandevices@mail.com

Lukasz-Juranek, 2019-10-21:
On other pieces this issue did not occurred. So most likely it was an single hardware issue.

closed 2019-10-21 by Lukasz-Juranek